alanesq/CameraWifiMotion #1: Save to SD card

paulohlcosta, 2020-02-04:
This is a very useful project, I am following it and would like see it working!

alanesq, 2020-02-04:
I believe the save to sd card feature is already working?
At the moment it saves a jpg file and a separate text file with time/date info., I need to look in to finding a better way of doing this?
btw-if using sd card you can not use the flash (this is a limitation of the hardware as the sdcard uses the led control i/o pin).

paulohlcosta, 2020-02-04:
Nice! I haven't heard about this flash/SD limitations. 

Yes, I would like to have it saved in the SD card, pictures with a 1000 ms interval. This guy made it save, seems easy, but his code has no motion detection.

https://randomnerdtutorials.com/esp32-cam-take-photo-save-microsd-card/

My current code:

alanesq, 2020-02-04:
The flash is on gpio pin 4 which is also used for the sd card so if your code is trying to turn the flash on this could account for your errors on the sd card.
I have some code in my setup which tries to load the sdcard and if successful sets a flag which tells the rest of the code not to try and use the flash.

paulohlcosta, 2020-02-04:
Sorry, here's the link to the code:
https://github.com/paulohlcosta/esp32-cam_motion_detection_sdcard/blob/master/esp32-cam_motion_detection_sdcard.ino

closed 2020-02-07 by alanesq

boontherd12346, 2020-03-13:
#4