alanesq/CameraWifiMotion #3: i2c driver install error

IncalSchorsch, 2020-02-23:
Hi - I'm glad having found your project. I plan to use it for watching my house heating system, which occasionally stops working. My plan is that I get an email when the error lamp is on.

So far your project covers a lot what I need, but I got a question about error messages I get whenever the camera shall take a picture. 
```
E (55614) i2c: i2c driver install error
E (56767) gpio: gpio_install_isr_service(410): GPIO isr service already installed
```
Everything else seems to function i.e. the webserver, the motion detection, email, etc.,, but I do not want to possibly operate it wrongly.

Do you have the same messages? Shall I worry about them?

I use ESP32 CAM Module with Arduino 1.8.12 and following settings: 
ESP32 Dev Module | 921600 | 240MHz | 80MHz | QIO | Flash 4MB | Huge App | Psram enabled

Cheers, Schorsch

alanesq, 2020-02-23:
I suspect it is because you have it set to "huge App"? as this allocates zero space for OTA.
Can you try it with it set to "default 4mb with Spiffs" and see if you still get the errors?

IncalSchorsch, 2020-02-28:
Good idea. I will try different settings when I am back home next week. 

I suspect it is because you have it set to "huge App"? as this allocates zero space for OTA.
Can you try it with it set to "default 4mb with Spiffs" and see if you still get the errors?

alanesq, 2020-03-08:
assuming all ok now

closed 2020-03-08 by alanesq