alanesq/CameraWifiMotion #5: Some errors

asjdf, 2020-03-12:
> Initialising camera: OK
> Updating bootlog: Booted
> 01-01-1970_00-01-24_Thu - Camera detected motion: 69
> Saved grey image to sd card
> E (87022) i2c: i2c driver install error
> E (88224) gpio: gpio_install_isr_service(410): GPIO isr service already installed
> Camera mode switched ok
> Taking a photo... attempt #1
> The picture has been saved as /1.jpg - Size: 46208 bytes
> Saved image to sd card
> E (90086) i2c: i2c driver install error
> E (91239) gpio: gpio_install_isr_service(410): GPIO isr service already installed
> Camera mode switched ok
> 01-01-1970_00-01-51_Thu - Camera detected motion: 90
> Saved grey image to sd card
> E (114311) i2c: i2c driver install error
> E (115514) gpio: gpio_install_isr_service(410): GPIO isr service already installed
> Camera mode switched ok
> Taking a photo... attempt #1
> The picture has been saved as /2.jpg - Size: 56960 bytes
> E (118312) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
> E (118312) diskio_sdmmc: sdmmc_write_blocks failed (263)
> 01-01-1970_00-01-56_Thu - Error: failed to save image to sd card
> E (123317) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
> E (123317) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
> E (123318) diskio_sdmmc: sdmmc_write_blocks failed (263)
> E (123325) i2c: i2c driver install error
> E (124472) gpio: gpio_install_isr_service(410): GPIO isr service already installed
> Camera mode switched ok
> 
> Sending NTP packet to uk.pool.ntp.org
> Seconds since Jan 1 1900 = 3792983285
> Too soon to re-trigger
> Too soon to re-trigger
> 12-03-2020_06-28-06_Thu - Camera detected motion: 48
> E (132571) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
> E (132571) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
> E (132572) diskio_sdmmc: sdmmc_write_blocks failed (263)
> E (137578) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
> E (137578) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
> E (137579) diskio_sdmmc: sdmmc_write_blocks failed (263)
> E (142584) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
> E (142584) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
> E (142585) diskio_sdmmc: sdmmc_write_blocks failed (263)
> 12-03-2020_06-28-22_Thu - Error: writing grey image to sd card
> E (142596) i2c: i2c driver install error
> E (143793) gpio: gpio_install_isr_service(410): GPIO isr service already installed
> Camera mode switched ok
> Taking a photo... attempt #1
> The picture has been saved as /3.jpg - Size: 41728 bytes
> E (150460) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
> E (150460) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
> E (150461) diskio_sdmmc: sdmmc_write_blocks failed (263)
> E (155467) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
> E (155467) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
> E (155468) diskio_sdmmc: sdmmc_write_blocks failed (263)
> E (160473) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
> E (160473) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
> E (160474) diskio_sdmmc: sdmmc_write_blocks failed (263)
> 12-03-2020_06-28-40_Thu - Error: failed to save image to sd card
> E (160487) i2c: i2c driver install error
> E (161646) gpio: gpio_install_isr_service(410): GPIO isr service already installed
> Camera mode switched ok

it have saved 3 pictures in sd card

alanesq, 2020-03-12:
it looks like it is having trouble writing to your sd card.
Have you tried a different sd card see if it is the same?

asjdf, 2020-03-12:
i have tried another program and the card is workable.i only have this tfCard so i can't do more test
it is 16GB Kingston tf card

asjdf, 2020-03-12:
when i trying to copy your code to my test program,i have another error,and i can't solve it.would you mind to help me solve this error?

> RestartCamera

> E (6980) ledc: freq_hz=0 duty_resolution=2
> [E][xclk.c:25] camera_enable_out_clock(): ledc_timer_config failed, rc=102
> E (6982) i2c: i2c driver install error

[This is the code i am testing](https://github.com/asjdf/ESP32CAMmotion_detection/blob/master/camChangeModeTest/camChangeModeTest.ino)

alanesq, 2020-03-12:
I have just tried an sd card in mine and still seems to work ok for me.
First thing I would check is your power supply.  If you do not have a good solid 5volts at 500ma and a decent smoothing capacitor you can get all sorts of weird things happening (especially if you have the flash enabled)

boontherd12346, 2020-03-13:
GiitHub.ok

asjdf, 2020-03-13:
> 
> 
> I have just tried an sd card in mine and still seems to work ok for me.
> First thing I would check is your power supply. If you do not have a good solid 5volts at 500ma and a decent smoothing capacitor you can get all sorts of weird things happening (especially if you have the flash enabled)

so... my code is work in your esp32? i just bought a new esp32 today. i will try it again and add a capacitor to the circuit.

boontherd12346, 2020-03-13:
Ok...

alanesq, 2020-03-28:
assuming all ok

closed 2020-03-28 by alanesq