alanesq/CameraWifiMotion #7: RebootCamera(pixformat_t format) & RestartCamera(pixformat_t format)

lbacksl, 2020-04-26:
Hello.
Nice project, thanks for sharing.
I try make similar project, but meet with problem change mode of camera PIXFORMAT_JPEG or PIXFORMAT_GRAYSCALE.
In the official repository esp32-cam is opened issue: [esp_camera_deinit() not working ](https://github.com/espressif/esp32-camera/issues/54)

I try Your code. At first it work normal but whent motion detected i get cyclic error:
E (38250) i2c: i2c driver install error
[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004
Updating bootlog: Camera failed to restart so rebooting camera
25-04-2020_22-32-51_Sat - ERROR: Problem with camera detected so resetting it

Please can you little explain how work change camera mode? How much time need to the switch?
I see in project espressif esp32-camera driver library. You used it directly in arduino code ? If yes how correct attache this library to arduino project? Default Arduino IDE use arduino-esp32 library.  
Thanks a lot.

alanesq, 2020-05-01:
Hi,
It may be worth uploading my bin file to see if this works to narrow down what the problem is.
i.e. If it still has this problem then the issue is with your hardware, if it works ok then the issue is with your IDE/compiler?
   https://github.com/alanesq/CameraWifiMotion/CameraWifiMotion.ino.esp32.bin

lbacksl, 2020-05-01:
Hello. 
Yes You are right. The problem was hardware. 
There are two cameras that are similar visually, but work differently:
OV2640 (marking TY-OV2640-v2.0) working perfect (right on photo).
XRZ00D1-V240 sensor which working with problems (left on photo).
[image]
[image]

Be carefully when buying on aliexpress.
Thanks.

closed 2020-05-03 by alanesq

move-some, 2020-05-12:
Hello
I have the exact same camera and the exact same failure.
Is there any way to solve this by changing something in the code?

Thanks