alanesq/CameraWifiMotion #8: Any possibility the get it working by modify the settings?

move-some, 2020-05-19:
@alanesq 

Hi Alan,
Is there any chance to modify the code to get the cam (in the link) working?

I have the exact same hardware and the same message in the log.

Thank you in advance

Steffen

https://github.com/alanesq/CameraWifiMotion/issues/7#issue-607007774

alanesq, 2020-05-19:
Hi,
I do not have one of the other cameras so am unable to try
You could try replacing the settings in "motion.h" with one of the cameras below?

---------

#if defined(CAMERA_MODEL_WROVER_KIT)
#define PWDN_GPIO_NUM    -1
#define RESET_GPIO_NUM   -1
#define XCLK_GPIO_NUM    21
#define SIOD_GPIO_NUM    26
#define SIOC_GPIO_NUM    27

#define Y9_GPIO_NUM      35
#define Y8_GPIO_NUM      34
#define Y7_GPIO_NUM      39
#define Y6_GPIO_NUM      36
#define Y5_GPIO_NUM      19
#define Y4_GPIO_NUM      18
#define Y3_GPIO_NUM       5
#define Y2_GPIO_NUM       4
#define VSYNC_GPIO_NUM   25
#define HREF_GPIO_NUM    23
#define PCLK_GPIO_NUM    22

#elif defined(CAMERA_MODEL_ESP_EYE)
#define PWDN_GPIO_NUM    -1
#define RESET_GPIO_NUM   -1
#define XCLK_GPIO_NUM    4
#define SIOD_GPIO_NUM    18
#define SIOC_GPIO_NUM    23

#define Y9_GPIO_NUM      36
#define Y8_GPIO_NUM      37
#define Y7_GPIO_NUM      38
#define Y6_GPIO_NUM      39
#define Y5_GPIO_NUM      35
#define Y4_GPIO_NUM      14
#define Y3_GPIO_NUM      13
#define Y2_GPIO_NUM      34
#define VSYNC_GPIO_NUM   5
#define HREF_GPIO_NUM    27
#define PCLK_GPIO_NUM    25

#elif defined(CAMERA_MODEL_M5STACK_PSRAM)
#define PWDN_GPIO_NUM     -1
#define RESET_GPIO_NUM    15
#define XCLK_GPIO_NUM     27
#define SIOD_GPIO_NUM     25
#define SIOC_GPIO_NUM     23

#define Y9_GPIO_NUM       19
#define Y8_GPIO_NUM       36
#define Y7_GPIO_NUM       18
#define Y6_GPIO_NUM       39
#define Y5_GPIO_NUM        5
#define Y4_GPIO_NUM       34
#define Y3_GPIO_NUM       35
#define Y2_GPIO_NUM       32
#define VSYNC_GPIO_NUM    22
#define HREF_GPIO_NUM     26
#define PCLK_GPIO_NUM     21

#elif defined(CAMERA_MODEL_M5STACK_WIDE)
#define PWDN_GPIO_NUM     -1
#define RESET_GPIO_NUM    15
#define XCLK_GPIO_NUM     27
#define SIOD_GPIO_NUM     22
#define SIOC_GPIO_NUM     23

#define Y9_GPIO_NUM       19
#define Y8_GPIO_NUM       36
#define Y7_GPIO_NUM       18
#define Y6_GPIO_NUM       39
#define Y5_GPIO_NUM        5
#define Y4_GPIO_NUM       34
#define Y3_GPIO_NUM       35
#define Y2_GPIO_NUM       32
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     26
#define PCLK_GPIO_NUM     21

#elif defined(CAMERA_MODEL_AI_THINKER)
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27

#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

------------

move-some, 2020-05-19:
Thank you. I will try one after the other. I already played with this settings but without success. Well, I'm no dev and an absolute Beginner, so hopefully, I do this in the right way.

mrbigglesw0rth, 2020-06-17:
same issue for me.
using CAMERA_MODEL_AI_THINKER

---------------------------------------
Starting - CameraWifiMotion - 11Apr20
---------------------------------------
SPIFFS mounted successfully.total bytes: 1378241, used bytes: 210840
01-01-1970_00-00-01_Thu - Loading settings from Spiffs
E (1585) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
01-01-1970_00-00-01_Thu - SD Card not found
Connecting to wifi
After waiting 0 secs more in setup() connection result is: 
 connected. Local IP: 192.168.1.92
mDNS responder started ok

Sending NTP packet to uk.pool.ntp.org
Seconds since Jan 1 1900 = 3801386986
MDNS responder started
Starting web server
17-06-2020_13-49-46_Wed - CameraWifiMotion Started
Initialising camera: OK
Updating bootlog: Booted
Not enough consecutive detections
Not enough consecutive detections
17-06-2020_13-50-33_Wed - Camera detected motion: 61
**E (52426) i2c: i2c driver install error
E (53623) gpio: gpio_install_isr_service(410): GPIO isr service already installed**
Camera mode switched ok
Taking a photo... attempt #1
The picture has been saved as /5.jpg - Size: 33920 bytes
E (55249) i2c: i2c driver install error
E (56405) gpio: gpio_install_isr_service(410): GPIO isr service already installed
Camera mode switched ok
Not enough consecutive detections

Sending NTP packet to uk.pool.ntp.org
Seconds since Jan 1 1900 = 3801387047

alanesq, 2020-08-31:
Can you try uploading the "CameraWifiMotion.ino.esp32.bin" file on my github and see if you still get the same problem.  
In case the problem is something related to the software setup you have (this will just upload a compiled image straight to your camera).  
Also check you have a very good 5 volt supply to the camera (short leads, ideally with a good size smoothing capacitor) as these boards tend to have all sorts of weird behaviour if the power supply is not perfect.