alanesq/CameraWifiMotion #14: Support PlatformIO

quixony, 2021-09-22:
I would strongly prefer being able to compile this awsome project within PlatformIO. 
This also would have the benefit that libraries are handled automatically (per define) 

I am working on a addition of MQTT service in the current project and would have been glad to use VSCode (jumping between funktions and files). Hopefully I will be able to pull the changes soon.

The changes needed to also support PlatformIO should not be immense.

tichaonax, 2021-10-25:
> I would strongly prefer being able to compile this awsome project within PlatformIO. This also would have the benefit that libraries are handled automatically (per define)
>
> I am working on a [...]

How far are you with MQTT. I want to explore automatic configuration with Homeassistant

alanesq, 2021-11-20:
I have been giving PlatformIO a try and managed to get it to compile and have uploaded the modified files
So it should now work in both Arduino IDE or Platform IO
Please give it a try and let me know if it works, what needs changing etc. (I am very new to Platform IO so still learning)

alanesq, 2021-11-21:
After seeing this thread it prompted me to have a try of PlatformIO and I have now modified my sketch so it works with both Arduino IDE and PlatformIO.