alanesq/CameraWifiMotion #15: Loads of errors when trying to compile

NorthyIE, 2022-05-10:
Hi,

I can't get your code to compile at all  trying both the Arduino IDE or platformIO.
In platformIO I get all kind of errors, from redifinitions to items not declared in scope, etc.

Using the ArduinoIDE I mainly get "error: too many arguments to function 'String currentTime()'" in multiple files.
I have made no changes (except organizing the files in their respective folders required for platformIO).

Any chance you could have a look at this?

alanesq, 2022-05-17:
Hi,
I have tested it compiles ok for me then re-uploaded all files to make sure there are no errors in the github version.

note: I am using esp32 board manager version 3.5.0 as the newer version seems to have issues with SPIFFS causing the esp32cam to restart.

SETHSETH, 2022-05-17:
In file included from /Users/mfb/Desktop/CameraWifiMotion/CameraWifiMotion/**CameraWifiMotion.ino:179:0:**
/Users/mfb/Desktop/CameraWifiMotion/CameraWifiMotion/motion.h: **In function 'bool capture_still()':
motion.h:217:32: error: return-statement with no value, in function returning 'bool' [-fpermissive]
     if (DetectionEnabled == 2) return;**            // if detection is paused another process may be using camera
                                ^
Mehrere Bibliotheken wurden für "WiFi.h" gefunden
 Benutzt: /Users/mfb/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5/libraries/WiFi
 Nicht benutzt: /Applications/Arduino.app/Contents/Java/libraries/WiFi
Mehrere Bibliotheken wurden für "SD.h" gefunden
 Benutzt: /Users/mfb/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5/libraries/SD
 Nicht benutzt: /Applications/Arduino.app/Contents/Java/libraries/SD
 Nicht benutzt: /Users/mfb/Documents/Arduino/libraries/SD
exit status 1
**return-statement with no value, in function returning 'bool' [-fpermissive]**

NorthyIE, 2022-05-17:
I fixed the main issues with compiling on my fork:
https://github.com/NorthyIE/CameraWifiMotion

The main issues were that the wifi settings were defined in wifi.h, which is the same name as the library itself, so it looped.
I also had to fix some other issues (for example setting currentTime() instead of currentTime(1), etc.) but it is compiling fine for me now.

alanesq, 2022-05-18:
I thought maybe the issues are when using Arduino IDE (as I use PlatformIO) so I just tried it but it compiles fine for me?
The command "currentTime()" should default to 1 if none supplied

andystumpf, 2023-12-02:
@NorthyIE Can you share the repository? The link is broken and I don't see it on the list of your repositories.  Thank you!!!