alanesq/CameraWifiMotion #16: wifi.h:90:21: fatal error: TimeLib.h: No such file or directory

grantiago, 2022-05-15:
Arduino: 1.8.19 (Windows 10), Board: "AI Thinker ESP32-CAM, 240MHz (WiFi/BT), QIO, 80MHz"

In file included from G:\Users\grant\Documents\Arduino\CameraWifiMotion-master\CameraWifiMotion\CameraWifiMotion.ino:180:0:

Multiple libraries were found for "WiFi.h"

wifi.h:90:21: fatal error: TimeLib.h: No such file or directory

compilation terminated.

 Used: C:\Users\grant\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi

 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi

exit status 1

TimeLib.h: No such file or directory

Am I missing a file?

alanesq, 2022-05-18:
Hi,
It looks like you require the timelib library installing (paulstoffregen/Time@^1.6.1)

grantiago, 2022-05-18:
Thank you. 1.6.1 is the timelib library  I have installed. It appears my zip download from github was corrupt. I reloaded all the files. that error is gone. Excuse me for using this a help desk. I am very interested in this sketch. On to the next issue.

closed 2022-05-25 by alanesq