alanesq/CameraWifiMotion #17: motion.h:217:32: error: return-statement with no value

grantiago, 2022-05-18:
Arduino: 1.8.19 (Windows 10), Board: "AI Thinker ESP32-CAM, 240MHz (WiFi/BT), QIO, 80MHz"

In file included from G:\Users\grant\Documents\Arduino\CameraWifiMotion-master\CameraWifiMotion\CameraWifiMotion.ino:179:0:

G:\Users\grant\Documents\Arduino\CameraWifiMotion-master\CameraWifiMotion\motion.h: In function 'bool capture_still()':

motion.h:217:32: error: return-statement with no value, in function returning 'bool' [-fpermissive]

     if (DetectionEnabled == 2) return;            // if detection is paused another process may be using camera

                                ^

Multiple libraries were found for "SD.h"

 Used: C:\Users\grant\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\SD

 Not used: C:\Program Files (x86)\Arduino\libraries\SD

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\grant\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi

 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi

exit status 1

return-statement with no value, in function returning 'bool' [-fpermissive]

Thank you.

alanesq, 2022-05-18:
That line should be "if (DetectionEnabled == 2) return 0; "
so I think you may have an older version of the file?

grantiago, 2022-05-18:
Hi. I pulled it this morning. https://github.com/alanesq/CameraWifiMotion/blob/master/CameraWifiMotion/motion.h
217    ` if (DetectionEnabled == 2) return; `
I updated the file locally 
    `if (DetectionEnabled == 2) return 0;`
it makes it past there. Thank you.

alanesq, 2022-05-18:
Just checked and you are right, I re-uploaded all the files yesterday but it looks like this one didn't update for some reason - I can't win ;-)

closed 2022-05-25 by alanesq