alanesq/CameraWifiMotion #18: should main.cpp be in the  CameraWifiMotion/CameraWifiMotion/ folder ?

dallday, 2022-06-20:
I tried compiling this with the Arduino IDE  and it had loads of errors about duplicated entries etc.,

After some investigation I saw that the main.cpp redefined a lot that was the the CameraWifiMotion,ino 

I removed the main.cpp from the folder for the Arduino IDE and it seems to compile and work successfully.

Is the main.cpp used for one of the other compiler options ?

Thanks for a very interesting project 

David

faisal-saddique, 2022-11-05:
I guess main.cpp is used for platformio

alanesq, 2023-01-13:
Yes, I converted over to PlatformIO but now I believe its no longer supported so in the process of moving to the new Arduino IDE...
As you did, just rename main.cpp to CameraWifiMotion.ini   (I should have put this in the notes)

closed 2023-01-13 by alanesq