alanesq/CameraWifiMotion #19: Still problems compiling in Arduino IDE...

LucidEye, 2022-12-17:
This sketch will not compile in Arduino IDE. I have placed all the libraries in the proper folder in the sketch, and directed the IDE to look for them there. However, there are still many errors when compiling.  After compiling and uploading, the ESP does not complete booting, and never connects to the WiFi.

I've attached the errors that happen during compiling.

I would very much like to use this sketch for building a doorbell camera.  Can you please offer some help as to why it isn't compiling or running?

Thank you

In file included from /home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino:179:
/home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/motion.h:139: warning: "FRAME_SIZE_PHOTO" redefined
         #define FRAME_SIZE_PHOTO FRAMESIZE_SVGA
 
/home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/motion.h:27: note: this is the location of the previous definition
   #define FRAME_SIZE_PHOTO FRAMESIZE_XGA       // Image sizes: 160x120 (QQVGA), 128x160 (QQVGA2), 176x144 (QCIF), 240x176 (HQVGA), 320x240 (QVGA), 400x296 (CIF), 640x480 (VGA, default), 800x600 (SVGA), 1024x768 (XGA), 1280x1024 (SXGA), 1600x1200 (UXGA)
 
In file included from /home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino:179:
/home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/motion.h: In function 'bool setupCameraHardware()':
/home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/motion.h:126:12: warning: 'camera_config_t::<unnamed union>::pin_sscb_sda' is deprecated: please use pin_sccb_sda instead [-Wdeprecated-declarations]
     config.pin_sscb_sda = SIOD_GPIO_NUM;
            ^~~~~~~~~~~~
In file included from /home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino:114:
/home/lucid/.arduino15/packages/esp32/hardware/esp32/2.0.5/tools/sdk/esp32/include/esp32-camera/driver/include/esp_camera.h:117:87: note: declared here
         int pin_sscb_sda __attribute__((deprecated("please use pin_sccb_sda instead")));           /*!< GPIO pin for camera SDA line (legacy name) */
                                                                                       ^
In file included from /home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino:179:
/home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/motion.h:126:12: warning: 'camera_config_t::<unnamed union>::pin_sscb_sda' is deprecated: please use pin_sccb_sda instead [-Wdeprecated-declarations]
     config.pin_sscb_sda = SIOD_GPIO_NUM;
            ^~~~~~~~~~~~
In file included from /home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino:114:
/home/lucid/.arduino15/packages/esp32/hardware/esp32/2.0.5/tools/sdk/esp32/include/esp32-camera/driver/include/esp_camera.h:117:87: note: declared here
         int pin_sscb_sda __attribute__((deprecated("please use pin_sccb_sda instead")));           /*!< GPIO pin for camera SDA line (legacy name) */
                                                                                       ^
In file included from /home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino:179:
/home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/motion.h:126:12: warning: 'camera_config_t::<unnamed union>::pin_sscb_sda' is deprecated: please use pin_sccb_sda instead [-Wdeprecated-declarations]
     config.pin_sscb_sda = SIOD_GPIO_NUM;
            ^~~~~~~~~~~~
In file included from /home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino:114:
/home/lucid/.arduino15/packages/esp32/hardware/esp32/2.0.5/tools/sdk/esp32/include/esp32-camera/driver/include/esp_camera.h:117:87: note: declared here
         int pin_sscb_sda __attribute__((deprecated("please use pin_sccb_sda instead")));           /*!< GPIO pin for camera SDA line (legacy name) */
                                                                                       ^
In file included from /home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino:179:
/home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/motion.h:127:12: warning: 'camera_config_t::<unnamed union>::pin_sscb_scl' is deprecated: please use pin_sccb_scl instead [-Wdeprecated-declarations]
     config.pin_sscb_scl = SIOC_GPIO_NUM;
            ^~~~~~~~~~~~
In file included from /home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino:114:
/home/lucid/.arduino15/packages/esp32/hardware/esp32/2.0.5/tools/sdk/esp32/include/esp32-camera/driver/include/esp_camera.h:121:87: note: declared here
         int pin_sscb_scl __attribute__((deprecated("please use pin_sccb_scl instead")));           /*!< GPIO pin for camera SCL line (legacy name) */
                                                                                       ^
In file included from /home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino:179:
/home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/motion.h:127:12: warning: 'camera_config_t::<unnamed union>::pin_sscb_scl' is deprecated: please use pin_sccb_scl instead [-Wdeprecated-declarations]
     config.pin_sscb_scl = SIOC_GPIO_NUM;
            ^~~~~~~~~~~~
In file included from /home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino:114:
/home/lucid/.arduino15/packages/esp32/hardware/esp32/2.0.5/tools/sdk/esp32/include/esp32-camera/driver/include/esp_camera.h:121:87: note: declared here
         int pin_sscb_scl __attribute__((deprecated("please use pin_sccb_scl instead")));           /*!< GPIO pin for camera SCL line (legacy name) */
                                                                                       ^
In file included from /home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino:179:
/home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/motion.h:127:12: warning: 'camera_config_t::<unnamed union>::pin_sscb_scl' is deprecated: please use pin_sccb_scl instead [-Wdeprecated-declarations]
     config.pin_sscb_scl = SIOC_GPIO_NUM;
            ^~~~~~~~~~~~
In file included from /home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino:114:
/home/lucid/.arduino15/packages/esp32/hardware/esp32/2.0.5/tools/sdk/esp32/include/esp32-camera/driver/include/esp_camera.h:121:87: note: declared here
         int pin_sscb_scl __attribute__((deprecated("please use pin_sccb_scl instead")));           /*!< GPIO pin for camera SCL line (legacy name) */
                                                                                       ^
In file included from /home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino:181:
/home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/standard.h: At global scope:
/home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/standard.h:88:74: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
 void webheader(WiFiClient &client, char* adnlStyle = " ", int refresh = 0) {
                                                                          ^
In file included from /home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino:181:
[... 17 lines cut ...]
/home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino: In function 'void handleImages()':
/home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino:1374:76: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   webheader(client, "#stdLink:hover { background-color: rgb(180, 180, 0);}");                 // html page header  (with extra formatting)
                                                                            ^
/home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino: In function 'void handleImagedata()':
/home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino:1477:78: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     webheader(client, "td {border: 1px solid grey; width: 30px; color: red;}");                // add the standard html header with some adnl style
                                                                              ^
/home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino: In function 'void handleBootLog()':
[... cut ...]
                   ^
/home/lucid/Arduino/CameraWifiMotion-master/CameraWifiMotion/CameraWifiMotion.ino:2200:49: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     sendEmail(_emailReceiver, _subject, _message);
                                                 ^
Sketch uses 1422253 bytes (45%) of program storage space. Maximum is 3145728 bytes.
Global variables use 59080 bytes (18%) of dynamic memory, leaving 268600 bytes for local variables. Maximum is 327680 bytes.

alanesq, 2023-01-13:
Hi,
I have just tried it with Arduino IDE v2.0.3 with all libraries etc. up to date  and it compiles ok for me but OTA will not work like this.  Use standard development board with PSRAM enabled and "minimal spiffs with ota" in partition table
I will re-upload my sketch now, just in case I have changed something recently and forgot to update it.

LucidEye, 2023-01-13:
I'm not trying to use OTA, I am connected to the ESP32 with a USB-UART adapter cable.
This is the ESP32 camera module I am trying to upload to.
https://www.amazon.com/gp/product/B0948ZFTQZ/ref=ox_sc_saved_image_2?smid=A1VTL661FOEJB1&th=1

mallemang, 2023-04-22:
this guy fixed the compile problems:  https://github.com/NorthyIE/CameraWifiMotion